spacetelescope/mirage #802: Problem with create_catalog.for_proposal()

Eobard828, 2022-09-04:
Traceback (most recent call last):
  File "/Users/eobardthawne/Desktop/Prof. C.Martin/MIRAGE/GO1635.py", line 62, in <module>
    catalog_results = create_catalog.for_proposal(xml_file, pointing_file,
  File "/Users/eobardthawne/opt/miniconda3/envs/mirage/lib/python3.9/site-packages/mirage/catalogs/create_catalog.py", line 156, in for_proposal
    pointing_dictionary = create_basic_exposure_list(xml_filename, pointing_filename)
  File "/Users/eobardthawne/opt/miniconda3/envs/mirage/lib/python3.9/site-packages/mirage/catalogs/create_catalog.py", line 64, in create_basic_exposure_list
    info = yaml_generator.SimInput(input_xml=xml_file, pointing_file=pointing_file,
  File "/Users/eobardthawne/opt/miniconda3/envs/mirage/lib/python3.9/site-packages/mirage/yaml/yaml_generator.py", line 344, in __init__
    self.apt_xml_dict, self.xml_skipped_observations = get_observation_dict(self.input_xml, self.observation_list_file, catalogs,
  File "/Users/eobardthawne/opt/miniconda3/envs/mirage/lib/python3.9/site-packages/mirage/yaml/generate_observationlist.py", line 502, in get_observation_dict
    xml_dict = readxml_obj.read_xml(xml_file, verbose=verbose)
  File "/Users/eobardthawne/opt/miniconda3/envs/mirage/lib/python3.9/site-packages/mirage/apt/read_apt_xml.py", line 359, in read_xml
    exposures_dictionary = self.read_nirspec_mos(template, template_name, obs,
  File "/Users/eobardthawne/opt/miniconda3/envs/mirage/lib/python3.9/site-packages/mirage/apt/read_apt_xml.py", line 2324, in read_nirspec_mos
    dithertype = template.find(ns + 'DitherType').text
AttributeError: 'NoneType' object has no attribute 'text'

Above is the error reported. When I check the mirage/apt/read_apt_xml.py script, I found that the link assigned to the variable "ns" does not work. I don't know whether it is the reason for this issue.
Thank u!

bhilbert4, 2022-09-06:
Thanks for reporting this. It looks like for MOS observations where the Dither Type is set to None in APT, there is no DitherType entry in the xml file. For other instruments it is present but set to None. I'll try to put in a fix this week.

Eobard828, 2022-09-22:
Hi，
Once this problem has been fixed, will I receive a message from you, or do I need to update the Mirage on my computer?

bhilbert4, 2022-09-26:
I'll put a message here when the update is complete. You should get an email. Once that is done, you will have to update Mirage on your computer. Sorry I've been slow on this. I've been swamped with high priority work for the last couple weeks.

Eobard828, 2022-09-26:
No worries, thanks for the update.

PR #806 (2022-11-10): Handle MOS observations with no dither entry
This PR updates the APT xml reader to handle MOS observations where no dither is specified. In this case, rather than having a DitherType = None in the xml file, the DitherType entry is not present.

Resolves #802
closed 2022-11-10 by bhilbert4